YCPRadioTelescope/TelescopeVisualization #88: Update TelescopeController to remember the telescope's current position and have "move to" commands

YCPjsteck, 2021-03-10:
The telescope model is currently controlled by sending commands through the TelescopeController. The TelescopeController is currently only set up to receive commands to move a certain number of degrees along each axis. This means that if you want to move the telescope to a certain position, e.g. to a defined position 0, the TelescopeController doesn't know how to do that.

Where this causes issues is that the control room sends commands to **move to** a certain position, not commands to **move by** a certain amount.

Update the TelescopeController script to remember its current facing position and have a **move to** command instead of just the current **move by** commands.

pdnelson, 2021-03-10:
I think the TelescopeController's "move by" commands are fine; the Control Room does the calculation and sends the simulation the orientation to "move by." So, this part is working fine.

The problem is, however, that the Control Room needs the telescope's current position in order to make that calculation, so some functionality needs added that sends the control room the simulation's current position. Otherwise, the control room thinks that it's always at (0,0).

YCPjsteck, 2021-03-28:
This issue was a matter of me speculating as to the cause of #85, but after further investigation it looks like the script actually already keeps track of its current orientation, and the newly renamed TargetAzimuth/TargetElevation functions (previously SetZ/SetY) do in fact tell it to move to a certain location.

closed 2021-03-28 by YCPjsteck